jethornton/mesact #1: Vanishing Tool

JayPerez1, 2022-07-19:
Hi BigJohnT,

My friend and I have been working on flashing the Mesa 7192m board I recently purchased and had just discovered your MesaCT tool. We were pretty excited about using it and taking advantage of all the powerful features you had built in.

I installed it on my home system on Saturday to check it out and prepare for installing it on the new LinuxCNC system we've been building for an existing CNC Router. That machine is sitting in the Rochester Makerspace that we're members of. I went there the next afternoon to install it and was having trouble finding the files I had just downloaded the day before from your site. Was able to retrieve the necessary files from my home system and complete the install.

We went back to your site later that day and found your message that you had removed the repository. I sympathize with your frustration and understand why you took that action.

Please let me know how we can help promote your work and if we can somehow persuade you to restore the repository so everyone can continue to take advantage of all the countless hours and incredible effort you put in for the community.

Respectfully.

Jay

closed 2022-07-19 by jethornton

jethornton, 2022-07-19:
Bill,

Thanks for your support, it means a lot to me.

JT

kiki64, 2022-07-22:
I have certainly been using this tool as well.  I am working on my first hobby lathe retrofit with my Dad and this tool has been very helpful while I am learning how Linuxcnc does things.  We finally have the kinematic parts, that turned out to be broken, replaced.  So I will be using this tool a bunch while we try to make a config(s?) that supports manual and CNC with a door / enclosure interlock.

Honestly it has been the one piece of software that worked on Debian Bullseye and Bookworm (after the fix).  The rest of Linuxcnc has been problematic for me on those versions so far.  Python3 and Wayland issues to name a couple.  Debian Buster is a no go for me due to it's End of Life being next month.

When I was deciding on electronics replacements I went with the Mesa 7i96s because I saw your tool had at least someone make support for it while I knew nothing about Linuxcnc and it's config files.

I have to fork and pull the latest version, but I think I have some enhancements that I'll post as issues.  I haven't written python in a while so I'm going to try submitting pull requests / suggestions if I can figure it out.

jethornton, 2022-07-23:
Thank you for the kind words and feedback.

Just curious why you had to fork the repo? 

JT

kiki64, 2022-07-23:
> Just curious why you had to fork the repo?

Forking allows me to have a online copy that I can try out, edit code, upload, and then submit pull requests upstream to your repo.  After that you can either merge it, edit, or close the pull request.

For instance, if I make a feature that isn't something you would put in the tool then I can still have it in a separate branch while contributing and merging other code / fixes in my main branch.

It essentially just allows for an easier way to contribute instead of posting code as a blob of text that can be hard to follow and review.  With Issues most of the work is on the developer.  Pull requests provide an example for the developer to either accept or start making their changes from there.

jethornton, 2022-07-23:
Thanks so much for wanting to contribute to mesact, I appreciate it very much and look forward to your pull requests.

Thanks
JT

JayPerez1, 2022-10-11:
> John,
>
> I also found your tool helpful and was just beginning to use it. Thank you
> for all of the time you spent developing it.
>
> Bill

Hi Bill !

Don't forget to add a comment in John's Github Issues section where others
can see it.

JayPerez1, 2022-11-17:
> Thank you for using my tool, knowing that at least one person uses it
> makes the thousands of hours worth while.
>
> JT

Hi JT,

Took many months but I finally have a PC running LinuxCNC 2.8.2 and my
friend and I have been able to start using your configuration tool. Very
nice !

We see updates every day on your GitHub account and are amazed at the
amount of effort and progress you've made !

I have a number of questions but not sure how you prefer to communicate.

Would you mind if we email you ?

Thanks again for all of your work on our behalf.

Jay

Vorms, 2022-11-17:
Hello,
I use your tool too !
Ï have not time available now, but I tried to make a config with a 7I97
card.
The result is the network of the machine full down...
It was an older version of your tool.
I can send you config files if you want.

Best regards
Thierry Vorms

commit 3a828f96 (2022-07-19): restore repo at users request fixes #1