jethornton/mesact #6: 7i96 I/O confusion

Pidgeonhawk, 2022-07-29:
Sir,
Thank you for this tool, it has been a great help in learning to configure my machine.

My question/issue has to do with configuring outputs on a 7i96 board using mesact.

When I map any of the 6 available OUTPUT pins on IO tab of mesact they are mapped sequentially from hm2_7i96.0.gpio.000.out to hm2_7i96.0.gpio.005.out.

My understanding is that on the 7i96 board, gpio.<IONums> 000 -> 010  are dedicated inputs on connector TB3 pins 1 -> 11. The outputs are gpio.<IONums> 011-016.

Am I misunderstanding how IOs are mapped (are all inputs AND outputs mapped from .000)?
If the outputs should be mapped from .011 and I need to hand edit the appropriate hal files, checking the 7i96_pins.txt file published on the info page of the code tree, I find that hm2_7i96.0.gpio.011.out -> mh2_7i96.0.gpio.016.out pins are not listed.

I do find the hm2_7i96.0.ssr.00.invert-xx and hm2_7i96.0.ssr.00.out-xx pins, but my logic wants to drive the outputs with the appropriate gpio.0xx.out pins.

Any feedback would be appreciated.

BR,
Rodney

jethornton, 2022-07-30:
Thanks for the report. 

Hang in there, I'm building a new development PC at the moment and should be able to look into this ASAP

JT

pcw-mesa, 2022-07-31:
The isolated outputs on a 7I96 can only be driven by the SSR hal pins

jethornton, 2022-08-01:
Thank you for your patience, I just pushed the fix for this error.

JT

closed 2022-08-01 by jethornton

Pidgeonhawk, 2022-10-11:
Cool, Thanks!

commit bc35a7f6 (2022-08-01): fix 7i96 output pin name fixes #6