jethornton/mesact #11: Program error

sprintertrd, 2022-08-24:
Installed mesact on a Linux CNC install from the linux CNC forum using  debian linux unstable sid X86-64 and version 2.9 linux CNC.
When I run mesact I get the mesa configuration too running however I also get a "program error occured in startup.py line 17 list index out of range" info box appearing.

Pressing OK closes the software.

jethornton, 2022-08-24:
Do you have LinuxCNC installed? When did you download the deb? I did some fixes 23 hours ago...

JT

jethornton, 2022-08-24:
I see what the error is now, when LinuxCNC is not installed there is an error in my code...

JT

closed 2022-08-24 by jethornton

jethornton, 2022-08-24:
Jay,

I'm the only programmer...

JT

JayPerez1, 2022-08-25:
Hi JT,

Thanks for the clarification. I've seen a number of emails from you and at
least one other person related to updates on your project and don't really
understand everything being said.

I've been struggling to install the latest stable version of LinuxCNC so I
can continue working on getting my Mesa board going. I've tried to install
it on three different computers and no luck.

I installed LinuxCNC 2.7 with Debian 9 Wheezy last night with no problem
and then upgraded LinuxCNC per the instructions. Had intended to upgrade to
2.8 but looks like it went to 2.9.

Not sure I want to test using this configuration but can't get my network
working with Debian 10 and can't get my graphics card to work with Debian
11.

Bill is going to try installing the current version on his computer at his
house as we wonder/suspect it might be something to do with my ISP and
DHCPv6 as that's where the installation hangs up.

Was on the LinuxCNC.org site last night and saw you listed as one of the
developers. You might want to check it out as there was a question mark
regarding your first name.

Thanks again for all your hard work and talent.

Jay

jethornton, 2022-08-25:
Have you seen my LinuxCNC installation guide?

https://gnipsel.com/linuxcnc/index.html

JT

JayPerez1, 2022-08-26:
I have not seen this before. Thank you for forwarding it.

Bill has a Mesa 7i96 but I'm using their 7i92m. Does that mean I don't have
to use the 2.9 Uspace install ?

commit 5db3bb39 (2022-08-24): fix version error when lcnc is not installed fixes #11