jethornton/mesact #13: Multiple Joint Axis   must be negative for at least one Joint.

rrln, 2022-09-09:
Hi,
the "Check config" function complains about:
"Multiple Joint Axis   must be negative for at least one Joint."

I have configured all input field s for all 5 axes (7i76e board), XYZAB, it still complains about Multiple Joint Axis.
There are no Multiple Joint Axis (XYZAB) so I don't understand the message. 
Any idea how to fix this? User error or sw problem?
Tried on current version.

BR Roger L

jethornton, 2022-09-09:
Probably a bug, I'll check it out in the morning.

jethornton, 2022-09-09:
Attach your ini file for me.

rrln, 2022-09-09:
[millton_2.9.ini.tar.gz](https://github.com/jethornton/mesact/files/9539004/millton_2.9.ini.tar.gz)
No hurry, I have a working setup since many years, attached ini file here.
Upgraded to 2.9 and are now interested in testing different setups/configurations.
Your tool seems to be very good for that.

And by the way, it complains about not able to find the pncconf file on ini file open.
So the ini is newer loaded. I supposed that it was possible to create a new ini file from scratch, I might be wrong here.

jethornton, 2022-09-09:
Can you attach the actual error message? The only time it complains about pncconf is if you try and open a pncconf file with mesact...

rrln, 2022-09-09:
[millton_2.9.pncconf.tar.gz](https://github.com/jethornton/mesact/files/9539091/millton_2.9.pncconf.tar.gz)
Ok, first, here is my pncconf file in case of...
The was originally created for many years ago. It is now upgraded to 2.9 and is edited and saved in 2.9 pncconf.
On opening the pncconf file in pncconf, I get a message:
"You have no hostmot2 firmware downloaded in folder: /lib/firmware/htm2/ PNCconf will use internal firmware".
I didn't bother since it works anyway in pncconf. 

On open ini file in mesact, I get a popup with heading "Incompatible File" and message "Can not open a PNCConf ini file".
I didn't see that before, "Incompatible File".

rrln, 2022-09-09:
[image]
Screenshot from "Open .ini File".

rrln, 2022-09-09:
Arg, I uploaded wrong files, they are manualy edited for 2.9.
I attach the files saved with pncconf 2.9 is attached in next post.
They should be the same, but have done any diff.

rrln, 2022-09-09:
[millton.tar.gz](https://github.com/jethornton/mesact/files/9539169/millton.tar.gz)
Same problem and message as for the other ini file. This has no XHC_HB04 config since I have not configure it in pncconf 2.9.
These files are the one edited in pncconf 2.9. The first post was only converted to 2.9.

jethornton, 2022-09-10:
Mesact does not use pncconf files.

closed 2022-09-10 by jethornton

rrln, 2022-09-10:
Ok, I did only open the ini file.

rrln, 2022-09-10:
I see that you closed this case, but I will let you know that I created a new working ini file from scratch with pncconf and I still get the "Incompatible File" and message "Can not open a PNCConf ini file". 
[my_LinuxCNC_machine.ini.tar.gz](https://github.com/jethornton/mesact/files/9540616/my_LinuxCNC_machine.ini.tar.gz)

rrln, 2022-09-10:
Aha, I can't use ini file created with pncconf at all. 
Then the original problem remains when creating a conf from scratch in mesact:
"Multiple Joint Axis must be negative for at least one Joint."
I'm not able to build a new config. 
FYI only, you don't have to dig deeper.
If you can send me a simple XYZA ini file working with mesact as a template I would be happy.

jethornton, 2022-09-10:
This has been fixed, you will have to download a fresh deb and install it to get the bug fix.

https://github.com/jethornton/mesact/commit/b8de5f14f1672415ff0d96644df01402c04ec87d

JT

rrln, 2022-09-10:
Great, many thanks! 
I will try it out tomorrow morning (Swedish time).

jethornton, 2022-09-10:
When I push a fix for an issue Github automatically closes the issue.

rrln, 2022-09-10:
Ok!

rrln, 2022-09-10:
Created some time for testing. :)
I can confirm that the fix worked. I'm able to create ini files and reload them in mesact. 

Now I have to figure out how to set i.e pos limit switch as shared with home switch.
And the analog spindle/dir signals.
I got them wrong in my first test, learning curve at my side and not related to this ticket,

I consider this ticket/issue as solved, thanks!

BR Roger L

commit b8de5f14 (2022-09-10): fix gantry checks fixes #13