jethornton/mesact #37: https://gnipsel.com/mesa/mesact/index.html is empty

Vorms, 2022-10-24:
unfortunatly, the link https://gnipsel.com/mesa/mesact/index.html is empty.
I would like to use the tool ton configure my very new 7I97 !
Tell me please some news !
Best regards
Thierry

tlight9, 2022-10-25:
Thierry, Try:  https://gnipsel.com/linuxcnc/mesact/index.html

Vorms, 2022-10-25:
Many thanks !

Regards

jethornton, 2022-10-25:
Redirect put in the old location.

closed 2022-10-25 by jethornton

Vorms, 2022-10-25:
Sorry ! I don't understand. Can you please explain to me ?
Best regards
Thierry

Vorms, 2022-10-25:
Hello,
Sorry to disturb you another time but I have followings error:
 mesact
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'

(atril:3210): GLib-GObject-WARNING **: 11:47:29.011:
g_object_get_is_valid_property: object class 'GtkSettings' has no property
named 'gtk-overlay-scrolling'
Empty filename passed to function
Empty filename passed to function

(atril:3992): GLib-GObject-WARNING **: 12:02:15.880:
g_object_get_is_valid_property: object class 'GtkSettings' has no property
named 'gtk-overlay-scrolling'
qt.qpa.xcb: QXcbConnection: XCB error: 3 (BadWindow), sequence: 5575,
resource id: 11198784, major code: 40 (TranslateCoords), minor code: 0

I am unable to write in any field.
I have a new 7I97 Mesa card. Do you support it ?

Many thanks for your help !

Best regards

Thierry Vorms