jethornton/mesact #38: Not possible to create a new config.
I can read existing config, but I have a message the config has not been created by mesact. I can't write machine name, etc because config was not loader. Is that possible to create a new config from scratch?
Other question: are planning to support 7i97 mesa card?

Regards
Thierry

Vorms, 2022-10-25:


jethornton, 2022-10-25:
I'll check in the morning and yes the 7i97 is supported.

JT

jethornton, 2022-10-25:
If your using master there may be bugs while I incorporate the new ini code, use 0.7.4 deb.

JT

Vorms, 2022-10-26:
Hello,
I successfully run the mesact. It's loock very nice. Many thanks for your work !
Question 1: I don't understand what is the difference between the axis scale and the encoder scale. For exemple, if I have 100 pulses of encoder for one milimeter, what is the value for axe scale and what is value for encoder scale ?
Question 2: I understand there are differents versions of the firmware. For me I have 3 linears axes and 2 spindles. I see I have the version 7I977D. Is that the good one ? is that the latest one ?
Question 3: How can I use the serial port in the mesa card ? what is the path ?

Many thanks for your replay
Regards

jethornton, 2022-10-27:
For LinuxCNC support ask on the forum or the IRC channel #linuxcnc.

JT

closed 2022-10-27 by jethornton