jethornton/mesact #56: Can't create a configuration with two daughter boards

JayPerez1, 2022-11-28:
Hi JT,

Not sure if this is the correct way to report issues or ask questions so please let me know if you prefer a different method.

I have a Mesa7i92m with two C10D breakout boards from CNC4PC.

I've successfully flashed the 7i92_5ABOB_Enc.bit firmware using your configurator.

**Issues:**

I'm only able to select a single 5ABOB daughter card on P2 on the Machine tab.

There are only 5 active drop-downs Inputs on the I/O tab (0-4). Inputs 5-15 and all of 16-31 are grayed out and not usable.

The GPIO tab has columns for Pins 1-17 but I cannot enter any values in the boxes and there are no drop-downs. These two issues have prevented me from defining all of my Homing/Limit switches in the ini and hal files.

Cannot disable Homing using the checkbox on the Options tab. Not sure this is a problem with your configurator as it doesn't seem to work setting the No_Force_Homing=1 variable in the ini file either.

I'm unsure of the proper way to define a slaved Y axis for my gantry. It sounds like you define a second Y axis in your configurator (XYYZ) and then just set the homing sequence to -2 on both of those axes. (Z=0, X=1, Y1=-2, Y2=-2) Is that correct ?

That's all for now.

Please let me know if there's any additional info you'd like or if I can be of help in some way.

Thanks for the great work and the great tool JT.

Jay

jethornton, 2022-11-28:
Thanks for the report, can you create a separate issue for each item? This way when I've addressed each item I can close the issue.

if  you select the 7i92_5ABOB_Enc.bit and look at P1 in the main window you will see it is all encoders. You would want the 7i92_5ABOB_x2D.bit for two 5 axis bobs.

I have not programmed the tool to do more than one daughter card yet but that's in the near future.

Yes, select as many joints as you have for that axis and make the home sequence the same for each one with at least one being negative.

Also note that F1 on any tab will display limited help information for that tab.

JT

JayPerez1, 2022-11-28:
Thanks JT,

Will do.

Jay

jethornton, 2022-12-03:
Right now I've only programmed for one daughter card. If you select one the other one gets disabled. Two daughter cards is in the works. On the Axes tab you can see the disabled Card 1 tab...

DrKnow65, 2023-01-19:
I am trying to configure a 7i98 + 7i76D + 7i85 combo that I have a firmware for provided by PCW on the forum. I can flash the firmware (after adding it to the 7i98 directory in libmesact) and then read the new HMID which has the correct number of muxedQcount, etc. But I cannot select even a single daughter board. 

I am able to flash the 7i98_7i77_7i74d.bit and have an option to select the 7i77 as a daughter board. 

Is there a place where I need to add firmware data to in order to get the mesact to populate one or the other of the daughter boards for even a partial config? 

Thank you for the new config tool!
Jeff