jethornton/mesact #58: Drop-down Inputs on I/O Tab

bweikertjr, 2022-12-03:
Hi JT,

Per your request I've broken out some of my questions/issues into separate reports.

There are only 5 active drop-down Inputs on the I/O tab (0-4).

Inputs 5-15 and all of 16-31 are grayed out and I cannot enter values into them.

jethornton, 2022-12-03:
The firmware sets how many I/O is available to use. Additional I/O will have to be freed up by reducing the number of step/dir, encoder or PWM functions. This also is a work in progress and would need to be somehow tied to the GPIO tab.

morlan, 2023-04-08:
Do the inputs on the I/O tab just coordinate with the Pins tab in the order listed in the firmware?  Any way that they can match?

[image]
[image]