jethornton/mesact #65: linuxcnc error

Vorms, 2022-12-19:
Hello,
I just generate a config with all default values and an Mesa 7I97 card
when I try to run linuxcnc I receive the followinf error.
Note: Using POSIX realtime
./conf1.hal:99: Pin 'hm2_7i97.0.pwmgen.00.enable' was already linked to signal 'joint-0-enable'
11617
Stopping realtime threads
Unloading hal components
Note: Using POSIX realtime

I can post all config files in pastebin if needed.
Thanks for your help !

jethornton, 2022-12-22:
Just attach the ini file to here.

Vorms, 2022-12-27:
Sorry,  I was outside.
I have to work for this project now.
Here is the ini file: https://pastebin.com/zSgipumP
Another question, I have modbus RS485 modules. Can I add these modules with your mesact software ?
Many thanks for your help. Best regards

Vorms, 2022-12-27:
As I understand the error produced by the linux cnc error:
Debug file information:
./conf1.hal:145: Pin 'hm2_7i97.0.pwmgen.00.enable' was already linked to signal 'joint-0-enable'

so, in conf1.hal (generated by masct:
# Spindle Board Connections
net spindle-on => hm2_[MESA](BOARD).0.pwmgen.00.enable
net spindle-vel-cmd-rpm hm2_[MESA](BOARD).0.pwmgen.00.value
setp hm2_[MESA](BOARD).0.pwmgen.00.scale [SPINDLE_0]MAX_RPM
setp hm2_[MESA](BOARD).0.pwmgen.pwm_frequency [SPINDLE_0](PWM_FREQUENCY)
setp hm2_[MESA](BOARD).0.pwmgen.00.output-type [SPINDLE_0](SPINDLE_PWM_TYPE)

SO, I think there is a troble with spindel config ?

Thanks for your help.
Thierry

jethornton, 2022-12-27:
Yes, the code that adds the spindle has a bug, I'll work on that in the morning.

Vorms, 2022-12-28:
Oh thank you a lot for your response.

Best regards
Thierry

jethornton, 2022-12-28:
This should be fixed now.

commit 004c0851 (2022-12-28): fix duplicate pwmgen used for spindle with analog boards
fixes #65
closed 2022-12-28 by jethornton