jethornton/mesact #70: https://github.com/jethornton/mesact/releases/tag/1.1.2

Vorms, 2022-12-29:
**Version and Build Date**
Version = 1.1.2
Build Date =29/12/2022

**Describe the bug**
./conf1.hal:149: Ini variable '[SPINDLE_0]PWM_FREQUENCY' not found. Mesa card 7I97
**To Reproduce**
Generate a config with a spindle

**Expected behavior**
To be able to have a spindle with pwm (tested with singl outpout)

**Config File**
ini file:
[SPINDLE_0]
SPINDLE_TYPE = analog
SPINDLE_PWM_TYPE = 0
MAX_RPM = 5000
MIN_RPM = 5

hal file:
# Spindle Board Connections
net spindle-on => hm2_[MESA](BOARD).0.pwmgen.03.enable
net spindle-vel-cmd-rpm hm2_[MESA](BOARD).0.pwmgen.03.value
setp hm2_[MESA](BOARD).0.pwmgen.03.scale [SPINDLE_0]MAX_RPM
setp hm2_[MESA](BOARD).0.pwmgen.pwm_frequency [SPINDLE_0](PWM_FREQUENCY)
setp hm2_[MESA](BOARD).0.pwmgen.03.output-type [SPINDLE_0](SPINDLE_PWM_TYPE)

**Operating System**
 OS =  Debian 10

**Additional context**

When I load the ini file, I receive a message:
No description file found for 7i97_D.bit

Many thanks for your work !
regards
Thierry

jethornton, 2022-12-29:
Many bit/bin files don't have a description file yet. Basically it's a text file with the stepgen/analog count and I/O count. 

Please attach your ini file.

Vorms, 2022-12-30:
OK I send you files by attachment
Best regards and many thanks
Thierry

jethornton, 2022-12-31:
This was a tough bug to fix but I found it...

closed 2022-12-31 by jethornton

Vorms, 2022-12-31:
Oh thanks a lot for your work |
I will test the new version when it will be disponible.
Have a nice day
Thierry

commit 6de4d54f (2022-12-31): stop updateini from deleting a key that starts with pwm fixes #70