jethornton/mesact #72: error on linuxcnc: invalidpwmgen and other errors with 7I97 card

Vorms, 2022-12-31:
**Version and Build Date**
Version = 1.2.0
Build Date =31/12/2022

**Describe the bug**
when I start the linuxcnc software, the dispay show errors:
hm2/hm2_7I97.0: invalid pwmgen output_type 0 requested

hm2/hm2_7I97.0 supported.output-type value are: 1(PWM & Dir), 2(Up &Down), 3 (PWM & DIr), and 4 (Dir & PWM)

hm2/hm2_7i97.0: switching to 1(PWM & Dir)

**To Reproduce**
Steps to reproduce the behavior:
1. Build files with mesact, choose 7I97 card

**Config File**
Attach the ini file that has the bug in it.

**Operating System**
 OS =  Debian 10

**Additional context**
Add any other context about the problem here.
Many thanks for your work. Have  nice day !
Regards

[config.zip](https://github.com/jethornton/mesact/files/10327894/config.zip)

jethornton, 2023-01-01:
Thank you for your testing and support.

JT

closed 2023-01-01 by jethornton

Vorms, 2023-01-01:
> via ef0b0d5

Hello,
I try today to reinstall the tool, but the same bugs remain.
Best regards
Thierry

jethornton, 2023-01-02:
Sorry, I got side tracked and forgot to upload the new deb

Vorms, 2023-01-02:
Hello,
Your new version seems to work ! Thanks a lot !
But the tool don't read the I/O from ini file.

Another question is which pin in the mesa card is corresponding to the
spindle ?

Thanks again and happy new year !

Thierry

jethornton, 2023-01-02:
Can you start a new issue? When you reply to a closed issue I get an email but sometimes I forget and if it not showing up in the repository then I might forget to fix it.

For and on/off spindle control any output can be used.

JT

commit ef0b0d5f (2023-01-01): fix pwm type combobox fixes #72