jethornton/mesact #74: The tool don't read the Inputs section

Vorms, 2023-01-02:
**Version and Build Date**
Version = 1.1.2
Build Date =13/31/2022

**Describe the bug**
the tool produce right file. But When it read ini file it doesn't read the inputs like limit switchs, home switch, etc.

**To Reproduce**
Steps to reproduce the behavior:
1. Build a config from scratch with I/O
2. Read the config

**Expected behavior**
The I/O must be read from the config

**Config File**
[INPUTS]
# DO NOT change the inputs they are used by the configuration tool
INPUT_0 = Joint 0 Home
INPUT_INVERT_0 = False

**Operating System**
 OS =  Debian 10

**Additional context**
Can you tell me the pin number for the spindle in the mesa card ?
Regards

jethornton, 2023-01-02:
Thank you for your continued input. I wish I had a 7i97 to test with.

You can use either an analog output or a digital output to control the spindle. The manual does not show a dedicated spindle interface.

jethornton, 2023-01-02:
fixed loading inputs and outputs

closed 2023-01-02 by jethornton