jethornton/mesact #77: Spindle configurations

Vorms, 2023-01-03:
**Is your feature request related to a problem? Please describe.**
I have a 7I97 card.
I can configure the spindle with two way: analog or digital.
I understand analog is with the use of a dac like an other axis on the card.
The 7I97 has 6 axis capability, I use only 3 so if I configure the spindle as analog I unserstand one of the axis analog output can be used for the spindle. Is tha right ? 
If yes, I have to match the spindle with one of analog output. Where can I do that ?
Or if I don`t understand, there is a pwm to fix the spindle speed. So, I have to make the integrater and etc. externaly.
But, I have to choose wich output is used for this function.
In the same context, It can be an on/off output 

If I choose digital spindle is that means the format for controling the spindle speed is digitial or is it an on/off control ?
If this is an on off control, I have to choose with output is used. In this case, there is no feedback and the encodor section can be disabled.

**Additional context**
I thank you very much for your time, the tool is now very usefull.
Sorry for my bad english....
An other suggestion can be to add a "debug section" to help with hardware for example:
- displaying input
- Comannd output
- Take in charge external file: I will have modbus inputs and outputs.

Best regards

Thierry

Vorms, 2023-01-08:
Hello,
Many thanks for your reply.
I have no problem with the hardware. I know where inputs and outputs and dac outputs are.
I don`t know how to configure the spindle to attribute a pin number if it is on/off or pwm if it is analog.

Another point is not clear for me.
I describe in mesact 3 axes. Join 0 for axe X and so on.
Is axe X linked with "out 0" and encoder 0 automatically ie without any file editing ?

Thanks for your help, thanks for your tool !
Best regards
Thierry Vorms

jethornton, 2023-01-08:
The joints use the same number encoder, pwmgen, stepgen etc.

Nothing is linked to any I/O automatically, they must be configured on the I/O tab.

Best place to ask about the 7i97 is on the forum.

closed 2023-01-19 by jethornton