jethornton/mesact #78: d/copyright - PNGs

smoe, 2023-01-04:
The PNGs describing the individuals cards, like https://github.com/jethornton/mesact/blob/master/mesact/src/images/7i95-card.png - are they yours and distributed under the same MIT license? d/copyright may otherwise need an update for these.

jethornton, 2023-01-06:
They are screen shots from the Mesa board manual.
http://www.mesanet.com/pdf/parallel/7i95man.pdf
Page 5

Looking at the 7i95 manual I don't see any copyright information.

JT

Vorms, 2023-01-08:
Hello,
Many thanks for your reply.
I have no problem with the hardware. I know where inputs and  outputs and
dac outputs are.
I don`t know how to configure the spindle to attribute a pin number if it
is on/off or pwm if it is analog.

Another point is not clear for me.
I describe in mesact 3 axes. Join 0 for axe X and so on.
Is axe X linked with "out 0" and encoder 0 automatically ie without any
file editing ?

Thanks for your help, thanks for your tool !
Best regards
Thierry Vorms

jethornton, 2023-01-08:
Thierry you replied to the wrong issue.

Vorms, 2023-01-08:
ok I copy past my message on the right issue. Sorry.
Thierry

jethornton, 2023-01-27:
The png files have been removed and the user can download them like the firmware files.