jethornton/mesact #80: configuration is not working with mesa 7i97 card

Vorms, 2023-01-10:
**Version and Build Date**
Version = 1.2.1
Build Date =10/1/2023

**Describe the bug**
The config produced by the tool is not usable

**To Reproduce**
Steps to reproduce the behavior:
Create a default config with 7i97 card

**Expected behavior**
working conf

**Config File**
[Attach the ini file that has the bug in it.
the files are attached
[01-10-23-09:25:09.zip](https://github.com/jethornton/mesact/files/10383531/01-10-23-09.25.09.zip) 
**Operating System**
 OS =  Debian 10

**Additional context**
If you haven't a mesa 7i97 card we can organize a meeting and you can connect to my machine using anydesk.
Best regard
I attached also a screen shoot 
[image]
Thanks for your work !

jethornton, 2023-01-12:
Axis loads ok? What is not working?
I do not have a 7i97 in stock to test with here.

jethornton, 2023-01-12:
At the very bottom of your ini file I see some duplicate section/key/values
[JOINT_0]HOME_SEQUENCE =  0
[JOINT_1]HOME_SEQUENCE =  1
[JOINT_2]HOME_SEQUENCE =  1

Did you add those?

Vorms, 2023-01-12:
Not at all.
As you can see on the screen capture, It's only axes X and Z. Both axes are
disabled.

Many thanks for your help. I can provide as many files and that you need.
Anydesk connetion is also possible.

Best regards
Thierry Vorms

jethornton, 2023-01-13:
That is the expected behavior when the E-Stop is on and power is off.

closed 2023-01-13 by jethornton

Vorms, 2023-01-13:
Hello,
Yes axis loads and work.
Yes you are right,, there was mistake added by me:
[JOINT_0]HOME_SEQUENCE = 0
[JOINT_1]HOME_SEQUENCE = 1
[JOINT_2]HOME_SEQUENCE = 1

I remove it and sorry for disturbing you with that.

But when I start linuxcnc with the conf, there is in the display only axe X
and Z but not the Y why ?

I can help you as you want.
Best regards
Thierry