jethornton/mesact #97: Library problem

zz912, 2023-03-08:
If I installed MesaCT, it has problem with library:
 
[image]

I solved it for myself easy. I copied library from Github to folder and it works.

jethornton, 2023-03-09:
When you have an issue first select Help > Check for Updates to see if you have the latest version. The current version is 1.2.3

closed 2023-03-09 by jethornton

zz912, 2023-03-09:
The problem was that when the 1.2.2 version wouldn't start, I couldn't go "Help > Check for Updates" to see that there was a 1.2.3 version.

jethornton, 2023-03-09:
I did forget to upload the latest html page...
https://gnipsel.com/linuxcnc/mesact/install.html

zz912, 2023-03-09:
There is still problem:
/1.2.2/mesact_1.2.3_amd64.deb - wrong
/1.2.3/mesact_1.2.3_amd64.deb - better