jethornton/mesact #103: Calculate Servo Thread

morlan, 2023-04-08:
Error message when clicking "Calculate" under both Servo Thread Test and NIC Test.
[image]

morlan, 2023-04-08:
Manually entered 2000 MHz in CPU Speed and Servo Thread Test Calculate works now but not NIC Test Calculate.
[image]

jethornton, 2023-04-08:
Can you copy and paste the error messages? I can't read them...

JT

jethornton, 2023-04-09:
Never mind I figured out how you got the first error and the second was fixed 2 weeks ago.

JT

commit 4ed4d0a3 (2023-04-09): change cpu speed to spin box fixes #103
closed 2023-04-09 by jethornton