jethornton/mesact #108: 7i96s + 7i78 not working

stav95, 2023-06-11:
**Version and Build Date**
Version = 1.2.4
Build Date =05/14/2023

**Describe the bug**
After selecting daughter card in Axes there is only 7i78 tab

**To Reproduce**
Steps to reproduce the behavior:

1. Selecting 7i96s
[image]

2. In Axes we can see 7i96s
[image]

3. Selecting my daughter card 7i78
[image]

4. The only tab available is 7i78
[image]

**Expected behavior**
I expected to see 2 tabs, one is 7i96s and the second 7i78

**Operating System**
 OS =  Debian 10

tonyluken, 2023-06-15:
I have the same issue with the 7i96s and a 7i77 daughter card - once the daughter card is selected, I can't do anything with the 7i96s. That includes not only the Axes tab but also the I/O and Spindle tabs as well.

jethornton, 2023-06-18:
Thanks for the report.

JT