jethornton/mesact #114: Step/Dir spindle decel issue

spumco1234, 2023-06-23:
**Version and Build Date**
Version = 1.2.4

Not a bug.  More like Mesact (_maybe)_ needs some tweaking on how a step/dir spindle config is generated.

To summarize, both PnCConf and Mesact create a step/dir spindle config with stepgen.N.enable connected to spindle.N.on.  I believe this connection results in instant decel which doesn't respect the max accel desired for the spindle.

I reported on the forum here:
[https://forum.linuxcnc.org/39-pncconf/49366-pncconf-mesact-step-dir-spindle-issue#273871](url)

**To Reproduce**
Steps to reproduce the behavior:
1. Create any config with a step/gen spindle
2. Set max accel for the spindle/stepgen
3. Check the resulting config.  spindle.N.on will be connected to stepgen.N.enable
4. When the spindle is turned off (gcode, halui, etc.) the stepgen will stop instantly.
5. If the spindle drive is in position mode with tight following error or gains, the spindle will abruptly stop - perhaps violently.

**Expected behavior**
Would like a basic config which respects the spindle accel/decel using a different HAL connection scheme

**Operating System**
 OS =  Debian 12

Not sure what the best solution is, given the various spindle on/off control methods and HAL schemes.  Probably only affects step/dir spindles with the drives in position mode (i.e. C-axis lathes with a shared spindle/axis motor)

Or I could be wrong about this... I don't have a working step/dir spindle (quite) yet, but I stumbled across various discussions of the issue while working on my current project.

jethornton, 2023-06-23:
interesting you posted that in the pncconf section which is only monitored by Chris and knowing him he will take offense to the mention of mesact in "his" section.

None the less an interesting problem to think about. A quick thought is spindle enable should be connected to power not spindle on.

JT

spumco1234, 2023-06-23:
Oh dear... didn't mean to cause offense.

It's not obvious who is 'in charge' of what in the LCNC backstage world.  I just dropped it in that sub as I noticed identical behavior between PnCConf and mesact.

If there's something I can do to help work on the issue let me know.

spumco1234, 2023-06-23:
And is what you mentioned one of the reasons qtpyvcp appears to have been (slightly) persona-non-grata'ed?

jethornton, 2023-06-28:
Yes Chris is the author of qtvcp and does not like pyqtvcp because the author worked on qtvcp with Chris until he realized that the approach was wrong.

jethornton, 2023-06-28:
I see Chris did not reply to your post...

JT

spumco1234, 2023-06-30:
I'm slightly surprised nobody's responded to my post.

Well, thanks for responding here.  I'm guessing you're knee-deep working on V2, but if/when you do get time to take a look at this issue I'll be happy to help with testing.

jethornton, 2023-06-30:
Ok, I'll see what I can do... it should be fixed in this and V2. You might ask a general question in Driver Boards about spindle decel, maybe Peter will chime in with some of his wisdom.

Aye, I'm plowing along with V2 trying to make it much better and easier to use than this one.

JT