jethornton/mesact #116: Angular velocities

persei802, 2023-06-27:
**Please use English for Bug Report**

**Version and Build Date**
Version = 1.2.3
Build Date =02/04/2023

**Describe the bug**
The check config tool reports:
Settings Tab:
	Default Angular Jog Velocity must be greater than zero
	Maximum Angular Jog Velocity must be greater than zero
However, those fields on the Settings tab are greyed out and it will not build the config file.
My axis configuration is XYZYA for joints 0 to 4, respectively.
All fields in the Axes tab are filled in.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Machine tab'
2. Click on 'Board'
3. Scroll down to '7i96'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect the angular velocity fields to be enabled so I can fill them in and generate a configuration file.
**Config File**
Attach the ini file that has the bug in it.
There isn't one.
**Operating System**
 OS =  Debian 11

**Additional context**
Add any other context about the problem here.

jethornton, 2023-06-27:
Thanks for the Issue Report but I can not reproduce that with the current Version 1.2.4 Build Date 08/14/23.

I selected 7i96 then on any joint select an angular axis this enables all the Angular Velocity spin boxes.

Always do Check for Updates from the Help menu first when you encounter an error or bug.

JT

persei802, 2023-06-27:
Thanks for the quick response. I updated to 1.2.4 and have to go through
the whole process again but I noticed it's not fetching the latest cache of
firmware files. The one I need is 7i96_pktdv2.bit but it's not listed in
the fetched firmwares. This means I can't use your tool to flash the board.
I find using mesaflash very cumbersome.
Jim

persei802, 2023-06-27:
BTW, how did you build this version 2 months in the future? :_)

jethornton, 2023-06-28:
Keyboard has a mind of it's own... the 8 is above the 5...

That firmware was only added very recently and I was not aware that any firmware has been added to the 7i96 which is no longer in production. The firmware repository has been updated to include the latest firmware from Mesa.

JT

jethornton, 2023-06-28:
Attach a copy of the ini file.

JT

persei802, 2023-06-28:
I tried but got "We don't support that file type" 
Try this.
[mesa.ini.tar.gz](https://github.com/jethornton/mesact/files/11900261/mesa.ini.tar.gz)

jethornton, 2023-06-28:
I'll have to try with Debian 11, all the angular boxes are enabled when I open your configuration in Debian 12. However I do see that the group box needs to be a bit wider...

[image]

jethornton, 2023-06-29:
[image]

[image]

I'm unable to reproduce this in Debian 11.

JT

persei802, 2023-06-29:
Not to worry. So long as it builds the ini file, even if there are errors in it. I need to make other edits to it anyway. BTW, I was looking through your code and I must say, that is some nice work. I may get some good ideas from it with my own work on QtDragon. You may consider this issue closed.

jethornton, 2023-06-30:
Thanks, take a look at mesact2 if your interested in the code base. I've made a lot of improvements there and that is my current project to finish it up and replace mesact with it.

JT

closed 2023-07-27 by jethornton